jwhitehorn/pi_piper #33: Increase EventMachine dependency version to 1.0.7

v0dro, 2015-04-10:
I'm facing some problems with building EventMachine 1.0.3 native extensions (a possible fix is given [here](http://stackoverflow.com/questions/17361958/eventmachine-gem-install-fail). 

Increasing the version to 1.0.7 (latest) fixes the issue.

v0dro, 2015-04-10:
And I think this issue arises only on ruby 2.2.0

jwhitehorn, 2015-04-10:
My schedule doesn't give me enough time to really work on this project at the moment, but I'd gladly accept a pull request.

v0dro, 2015-04-10:
Sure thing!

Also could you tell me if there is anything specific that I should keep in mind during testing?

elmatou, 2015-05-05:
Same here, happend only with ruby 2.2.x . Eventmachine 1.0.4 (a later) fix this.

elmatou, 2016-02-07:
EventMachine dependency could be withdrawn. 
It is only used for PiPiper#spi_poll which could be done with a simple sleep in a loop. Other Implementations, with EventMachine, Celluloid, Cool.io, Thread, Fiber.... should be done in application code.

zsyed91, 2016-02-08:
I agree with @elmatou. If we can move this into the actual app code then it should reduce complexity of the gem. But anyhow, the current changes going into `1.9.9` which hopefully should be released in the next day or so contains the latest version of `EventMachine`. So this issue should be resolved once that is released. I am going to go ahead and close this.

closed 2016-02-08 by zsyed91