jwhitehorn/pi_piper #37: non-root access

C0de-M0nkey, 2015-04-22:
Hi,

non-root access to the dev and sysfs devices seems to be a recurring issue going by the many threads all over the internet, but is there any change on the pi_piper front ? Or is there anyway to proxy access via a root daemon ?
I appreciate this question was answered a couple of years ago, just wondering if anything has changed since then.

Thanks

CM

elmatou, 2015-05-05:
PiPiper make use of gpio kernel wich use /sys/class/gpio hierachy which is owned by root...
bcm2835_functions branch make use of /dev/mem memory access which also need to be root...

As a workaround, I'm looking at two scenarii :
1 - run ruby app with sudo, fork a proc as root, and descalate permission on main thread.
2 - run as root a ruby daemon with DRb server and run a non-privileged processus as DRb client.

Anyhow, at some point root is needed, and will always be needed as far as I know

best regards

C0de-M0nkey, 2015-05-19:
Hi Elmatou, 
Thank you for your response. This more or less confirmed my thoughts. It seems like a root daemon would be the strategic solution in the long term. Anything else would seem to require reinventing the wheel for each project.

Thanks again.

CM