jwhitehorn/pi_piper #39: Update libbcm2835.img

elmatou, 2015-05-05:
We can see at http://www.airspayce.com/mikem/bcm2835/ the librairy is quite often updated, even more now with Rpi2 support.

I was willing to update the image embeded in the gem but I'm not able to build the correct image, I'm stuck with the library libbcm2835.a

What is the correct procedure ? would it possible to build a rake task on it (I can do it and send a PR) ?

thx for the help.

sean2121, 2015-05-27:
Hi,

Jury-rigged solution to use pi_piper on my Pi2.
Instead of update libbcm2835.img, I've installed the newest bcm2835 and use a libbcm2835.so 

Sean.

sean2121, 2015-05-27:
oops.

I didn't know that libbcm2835 has already updated in pi_piper-2.0.beta.11

elmatou, 2015-05-29:
Sorry I'm not very familiar with C compilation and type of library. 
If I understand well .so stands for shared object, meaning dynamicly linked library, as .a stand for static library. currently the Makefile of libbcm2825 create only a static library (.a file).
@sean2121, What is the process to build .so file.
@jwhitehorn, what is the image file embedded in the gem ?

elmatou, 2016-02-06:
@jwhitehorn, @zsyed91 any hint on how to update the library to the last version (1.49) ?

closed 2016-02-07 by elmatou