jwhitehorn/pi_piper #40: Running pi_piper in Rails

canedy, 2015-05-07:
Looking for some advise or if anyone has been able to get a basic blinking light scenario website up and running using pi_piper. I'm starting as simple and basic as I can. Created a rails app and in my controller added same ruby code (that works when ran as ruby). Was expecting page I navigated to see lights blink, nothing fancy yet just wanted the code to execute. But i received a Permission denied  error.

Anyone out there have experience with getting pi_piper to work with rails?

elmatou, 2015-05-08:
Hi,
PiPiper needs root permission to work, it access to /dev/mem (via a kernel module or directly wih the C library) which need root permission.

As it is not advised to run a webserver as root, I will recommend you to escalate permissions only for a PiPiper dedicated process, there is an issue about it, but nothing as been properly written as of now.

If you make any progress on this front keep us up to date.

bguest, 2015-05-08:
This is a site I wrote to do LED stuff: https://github.com/bguest/blinky... basically you need to deploy to root.

canedy, 2015-06-23:
So after much trial and error I end up installing everything as root. I could not get rails to write to GPIO pins with out installing ruby, rails, pi piper all as root. It works, but there has to be a better way other than installing as root.

closed 2015-06-23 by canedy