jwhitehorn/pi_piper #43: how to call a method from inside PiPiper.after?

joecabezas, 2015-06-13:
This looks more like a ruby question, but it is related to pi_piper. I have this code:

    require 'pi_piper'

    class Foo

        def initialize
                PiPiper.after ({pin: 7, goes: :high}) do
                        bar
                end
        end

        def bar
                puts "hello"
        end
    end

    a = Foo.new

    PiPiper.wait

but when I turn on the pin (with a switch), it returns this error

›sudo ruby stack.rb
stack.rb:7:in `block in initialize': undefined local variable or method `bar' for #<PiPiper::Pin:0x1172508> (NameError)
        from /var/lib/gems/1.9.1/gems/pi_piper-1.3.2/lib/pi_piper.rb:24:in `instance_exec'
        from /var/lib/gems/1.9.1/gems/pi_piper-1.3.2/lib/pi_piper.rb:24:in `block (2 levels) in watch'
        from /var/lib/gems/1.9.1/gems/pi_piper-1.3.2/lib/pi_piper.rb:19:in `loop'
        from /var/lib/gems/1.9.1/gems/pi_piper-1.3.2/lib/pi_piper.rb:19:in `block in watch'

how can effectively call "bar" method when pin goes high?, thanks

elmatou, 2015-06-13:
When calling `PiPiper.after` (or `wait` by th way) without argument, it call block in the scope of the `PiPiper::Pin` instance, where `bar` is not defined. 
try with an argument (which will be the pin instance): 
```ruby
PiPiper.after ({pin: 7, goes: :high}) do |pin|
  bar
end
```

joecabezas, 2015-06-13:
thank you for quick answer, I really appreciated, that made the trick :)

Can you tell me some keywords to google this problem and study more abut this?

I was looking for: "ruby blocks", "ruby instance_exec", "ruby calling method inside block"

but nothing useful, thanks in advance!

zsyed91, 2016-10-18:
Here is a good blogpost to anyone else who stumbles upon this issue: http://www.eriktrautman.com/posts/ruby-explained-blocks-procs-and-lambdas-aka-closures

closed 2016-10-18 by zsyed91