jwhitehorn/pi_piper #44: PWM? how to make a buzzer sound?

joecabezas, 2015-06-13:
Hello, I want to make a piezo buzzer sound, there is a way to do this?, is pi_piper able to pwm?, thanks

elmatou, 2015-06-13:
not the released gem, nor the develop branch. But their is two forks with this implementation : 
* mine using only the bcm library : https://github.com/elmatou/pi_piper
* another relying on piblaster, a binary allowing PWM software or hardware

joecabezas, 2015-06-15:
thanks, how to install your library?

elmatou, 2015-06-17:
not particularly a pi_piper question, use bundler ?

gem 'nokogiri', :git => 'https://github.com/tenderlove/nokogiri.git'

just change je the gem name and the github url.

elmatou, 2016-02-15:
As of today PWM develop branch support Hardware PWM in pin 18.
if everything is ok please close the issue

joecabezas, 2023-07-08:
Thanks everyone for making the community awesome <3, I am closing this

closed 2023-07-08 by joecabezas