jwhitehorn/pi_piper #45: /home/pi/.rvm/gems/ruby-2.2.1/gems/pi_piper-1.3.2/lib/pi_piper/pin.rb:23:in `close': Device or resource busy @ fptr_finalize - /sys/class/gpio/export (Errno::EBUSY)

galori, 2015-12-11:
Trying to use pi_piper to read a button connected to pin 18 and getting this error:

```
/home/pi/.rvm/gems/ruby-2.2.1/gems/pi_piper-1.3.2/lib/pi_piper/pin.rb:23:in `close': Device or resource busy @ fptr_finalize - /sys/class/gpio/export (Errno::EBUSY)
	from /home/pi/.rvm/gems/ruby-2.2.1/gems/pi_piper-1.3.2/lib/pi_piper/pin.rb:23:in `open'
	from /home/pi/.rvm/gems/ruby-2.2.1/gems/pi_piper-1.3.2/lib/pi_piper/pin.rb:23:in `initialize'
	from /home/pi/.rvm/gems/ruby-2.2.1/gems/pi_piper-1.3.2/lib/pi_piper.rb:18:in `new'
	from /home/pi/.rvm/gems/ruby-2.2.1/gems/pi_piper-1.3.2/lib/pi_piper.rb:18:in `block in watch'
```

This is my code:
```
$ cat pi_piper.rb
require 'pi_piper'
include PiPiper

pin = PiPiper::Pin.new(:pin => 18, :direction => :in, :pull => :up)

watch :pin => 18 do
  puts "Pin changed from #{last_value} to #{value}"
end
```

I'm running it like this:
```
$ rvmsudo bundle exec ruby pi_piper.rb
```

Conversely running this program in python works just fine (and outputs "Button Pressed" correctly only when I press the button)

```
pi@raspberrypi ~ $ cat switch.py
import RPi.GPIO as GPIO
import time

GPIO.setmode(GPIO.BCM)
GPIO.setup(18, GPIO.IN, pull_up_down=GPIO.PUD_UP)

while True:
  input_state = GPIO.input(18)
  if input_state == False:
    print('Button Pressed')
    time.sleep(0.2)

$ sudo python switch.py
Button Pressed
Button Pressed
Button Pressed
Button Pressed
Button Pressed
```

Any advice? Anything I can try?

jvlobo, 2015-12-22:
I'm having the same issue @galori 

Did you be able to solve it??

Thanks.

bmedici, 2016-01-10:
Same here... would be nice to know what it means !

elmatou, 2016-01-10:
Hi @galori,
the error 
`Device or resource busy @ fptr_finalize - /sys/class/gpio/export (Errno::EBUSY)`
means that the pin is already in use. 
Reading your code we can see you are the first user.
`PiPiper::Pin.new(:pin => 18...` reserves (exports) the pin 18 for your usage with the `pin` variable you create.
the `PiPiper.watch` method is a shortcut for `pin = PiPiper::Pin.new()` && `pin.wait_for_change()` on the created object, as you can see here : https://github.com/jwhitehorn/pi_piper/blob/develop/lib/pi_piper.rb#13
your code should only look like :
```ruby
watch :pin => 18, :pull => :up do
  puts "Pin changed from #{last_value} to #{value}"
end
```

ahendricks, 2016-01-13:
PiPiper doesn't release (unexport) the pins for me at the end of my programs.

To release the pin at the end of the script I do this, where 18 is the pin number:
`File.open("/sys/class/gpio/unexport","w") { |f| f.write(18) }`

colinl, 2016-01-20:
Would it be safe to use
File.open("/sys/class/gpio/unexport","w") { |f| f.write(pin_number) }
in my code before I use pin.new in order to protect against the problem of a pin not being released on a crash?

elmatou, 2016-01-20:
It could be safe inside your thread. but what if the pin is already in use
in another program or thread ? it would loose controle of the pin.

El Matou

colinl, 2016-01-20:
A good point elmatou, though if I have two processes or threads both trying to control a relay in my home automation system then I am in big trouble.  Is there any good way round the problem in general?  Ideally the pins should be released automatically if a process crashes, in the same way as happens with system resources (RAM etc), but I do not know whether that is possible.

elmatou, 2016-01-20:
that's why I'm trying to use the bcm2835 C library for gpio (pi_piper uses
it for SPI and so...)
It don't need any export/unexport procedure and work pretty well...

Except for the event-driven side. still working on it :
https://github.com/elmatou/pi_piper

El Matou

colinl, 2016-01-20:
Thanks El Matou, that sounds good. I will come across and take a look.

zsyed91, 2016-02-08:
To all in this thread:

We just released version `1.9.9` that fixes this issue. Pins are released automatically at program exit or can be manually released by `pin.release`.

avinash386, 2016-06-03:
pin.release is not working in 2.0.0 , can you suggest a work around .

zsyed91, 2016-06-03:
Can we get more info? Stacktrace and your code would help. But for starters are you running rvmsudo?

avinash386, 2016-06-03:
Yes , I am running the code as super user. I am trying to invoke a switch and try switching on a led. 

So as per your example code , when I use Pipiper.wait it runs a loop. On my 1st loop the led pin is getting high, on my second loop it's giving me an error as the GPIO pins are not unexported. 

I wish to light up the led every time it detects switching between states of the switch.

zsyed91, 2016-06-03:
Ok just to make sure,

`sudo ruby script.rb` is not the same as `rvmsudo ruby script.rb`. Running just `sudo` does not pass the permission level to the script. See the [rvmsudo documentation](https://rvm.io/integration/sudo).

Regarding the sample code, if you could just paste your actual code here that would help more. I cannot assume what your loop looks like.

avinash386, 2016-06-04:
require 'pi_piper'
include PiPiper

red     = PiPiper::Pin.new(:pin => 15, :direction => :out)
green   = PiPiper::Pin.new(:pin => 18, :direction => :out)

after :pin => 14, :goes => :low do
        red.on
        green.on
        sleep 1
        red.off
        green.off
        sleep 1
end

PiPiper.wait

Please find my program above. I wish to come out of the program after certain number of button activity. When i come out of my program before execution, the pins are still active. so i have to perform echo "18" > /sys/class/gpio/unexport to unexport them.

If there is any funtion that i can implement in my program to release my pins, that would be great.

zsyed91, 2016-06-15:
Did you try running `rvmsudo ruby your_script.rb`? I don't see anything obvious in the code.