jwhitehorn/pi_piper #48: State of project

zsyed91, 2016-01-28:
What is the state of this project? I haven't seen any commit in the past few months. I think its a great library and would love to see some movement. Especially because there are several pending issues and open PR's that would fix some issues everyone is having.

jwhitehorn, 2016-01-28:
It could use a maintainer, as I lack the time to devote to this project at the moment.

zsyed91, 2016-01-28:
Ah I see, completely understand. I am still fairly fresh to this project so I am not sure if I could take on that role. However, I would not mind if no one else wants to do it, and you feel like I can handle it. Some of the more technical stuff on the pi is a bit above me at the moment but I do understand how the pins work.

jwhitehorn, 2016-01-30:
Well, you can't do any worse than I have :-)

closed 2016-01-30 by jwhitehorn

zsyed91, 2016-01-30:
Haha alright. Suddenly I feel an increase in pressure :P
Hopefully I won’t disappoint. Looking forward to holding down the fort until you free up some time.

elmatou, 2016-01-30:
@zsyed91, I'll give you a hand and will make some proposals from my fork : https://github.com/elmatou/pi_piper

zsyed91, 2016-01-30:
@elmatou Awesome. You seem to have good experience with this library. Can you take a peek at my PR for upgrading rspec. If you think its good I can merge. I am going to take a look at the  PR for releasing the pins at exit and merge that in shortly.