jwhitehorn/pi_piper #54: Roadmap for PiPiper

elmatou, 2016-02-14:
@jwhitehorn, @zsyed91,
I'm willing to bring some enhancement to the library, I just need to know if we are ok with it !

* Better Licence : use standart licence (was a BSD?), add LICENCE file, add licence reference for the bcm2835 lib
* Add support for PWM
* Spec without a stub for Pin,I2C, ...
* Add support for multiple drivers (kernel module, bcm2835 lib, WinringPi, ...)
* Spec the drivers (how ?!)
* Non-root usage.
* ...

jwhitehorn, 2016-02-14:
I'd prefer to stay with a BSD license.

zsyed91, 2016-02-14:
Looks like a good plan to me! @elmatou I will review your PWM pr, sorry I have been busy but will get to it today.

elmatou, 2016-02-15:
@jwhitehorn BSD is a very good choice. I'll add a LICENCE file

elmatou, 2016-02-21:
Here are my thoughts for your comments o how could be implemented the multi drivers thing.
```ruby
# Platform handle the loaded drivers and the way to load new drivers.
class Platform
  autoload StubDriver, 'blabla.rb'
  autoload SysFs, 'blabla.rb'
  autoload Bcm2835, 'blabla.rb'
  ...

  @@drivers_pool = {}

  def self.link_with_driver(driver_klass)
    unless @@drivers_pool[driver_klass].instance_of? driver_klass
      @@drivers_pool << driver_klass.new
    end
    @@drivers_pool[driver_klass]
  end

  def self.unlink_driver(driver_klass)
    ...
  end

  def self.empty_pool
    ...
  end
end

# StubDriver used to define the standart API, and to stub for testing
class StubDriver
  def initialize
    at_exit { close }
    # or define a finalizer : ObjectSpace.define_finalizer(object, proc)
  end

  def close
    # unexport pin
    # close librairy
    # ...
  end

  def each_public_method(*args)
    # kept empty for test matter
  end
end

# each driver could be in a separate gem (or just not required from start)
class SysFs < StubDriver

  def initialize
    @exported_pins = []
    super
  end

  def close
    @exported_pins.each {|p| unexport_pins(p) }
  end

  # Standart public methods conform to the 'API'
  # Methods prefix with the object they are used for
  def pin_set_...(gpio_number, value)
  end

  def pin_read(gpio_number)
  end

  def spi_poll
  end
  ...
  
  # private methods are not intended to be called from outside the driver.
private
  def export
  end

end

# Other Drivers could be bcm2835, gordon's wiringPi lib, platform specific lib for beaglebone, ordroid.... 

class Pin
  def initialize(options)
    # driver can be choosed at initialization time. it could be different for each object (in order to gain controle of specific behaviour)
    @options[:driver] = Platform.link__with_driver options[:driver]

    # we vallidate the options by setting them through the driver. Settings cannot be changed later.
    # the driver raise exceptions for invalid arguments
  end

  def close
    # Optional : close procedure/release if we cannot wait for the programm to end or the GC to collect
  end

...

private
# Optional : a way to access the driver directly for driver specific behaviour
  def method_missing(method, *args, &block)
    @options[:driver].send(method, @options[:pin], *args, &block)
  end
end

class Spi
# Same here
private
  def method_missing(method, *args, &block)
    @options[:driver].send(method, @options[:pin], *args, &block)
  end
end
```

zsyed91, 2016-02-21:
I like the idea of having separate driver gems. I think along these lines we could even do the following:

```ruby
class Driver
  def close
    raise ArgumentError , 'NotImplemented'
  end
end

class StubDriver < Driver
end

class SysFs < Driver
end
```

Then we can have a basic `Driver` class that we would ducktype under and have the actual drivers implement the public api methods.

zsyed91, 2016-02-21:
@elmatou I have created a placeholder repo under a new organization `PiPiper` for the bcm2835 library. We can start creating a new gem there. And eventually same with the other drivers.

https://github.com/PiPiper/bcm2835

elmatou, 2016-02-21:
Funny, it was precisely the structure of my first draft. Then I asked me, why do we need StubDriver ?
The specs in PR #57 are unit tests, the do not need a driver t mock/mimic behaviour, only some thing to receive de methods call. That's why I proposed a Driver object with methods declarations whithout any code in it. Any how, I think we are in the write way.

elmatou, 2016-02-21:
the placeholder, nice !

I see, for me to work on :
* pi_piper
* pi_piper-sysfs
* pi_piper-bcm2835
* pi_piper-wiringpi
then will be for contributors
* pi_piper-beagle
* pi_piper-cubie
* pi_piper-banana
* pi_piper-.....

zsyed91, 2016-02-21:
Yeah I think we can use the StubDriver to make sure method calls are sent and received properly instead of actually testing the code itself. If we use a separate gem for each driver (or really even if we don't) we can stub the driver and unit test the code that goes in the driver separately in that repo to unit test the driver.

zsyed91, 2016-02-21:
Yes we can create more placeholders once we get this moving. I am going to go ahead and create a branch off of `develop` that we can push our structure changes to in the meantime. That way we can still continue to release the gem as needed while we do this work in parallel.

zsyed91, 2016-02-21:
Alright, I think we can start pushing any structure changes for this to this branch: `gem_structure_changes`

https://github.com/jwhitehorn/pi_piper/tree/gem_structure_changes

zsyed91, 2016-02-22:
@elmatou I copied over the existing `bcm2835` code here https://github.com/PiPiper/bcm2835/tree/dev
I think we can probably add your tests in the spec directory there, and begin to make improvements. And as for your current PR, can you close it and reopen it to `gem_structure_changes`? That way we can be a little more aggressive in our changes without worrying about affecting near term releases.

elmatou, 2016-02-22:
sh.t I was doing it offline... nevermind I'll work on your structure. let's work on gem_structure_changes
 branch (or something badass like, reborn or reboot !!)

elmatou, 2016-03-04:
I'm currently trying to implement the external driver concept for PiPiper, but it leads me to put arguments validation in the driver, because every drivers do not accept the same set.
eg : sysfs only support rising, falling, both, as bcm2835 offer async trigger and more

but I'm not very comfortable with it, as it will probably not be backward compatible.... any thoughts ?

zsyed91, 2016-03-04:
I think having driver specific validations are alright. As you said, if sysfs only supports `:rising, :falling, :both` but another driver supports only `:rising` for example then it should throw a validation error. This is the plus side to our gem approach for the drivers since they can all be stand-alone.

I think we don't have to worry about backward compatibility because we can release those changes under `3.0`. If people need to keep the old version then that is fine, they will just have to change their gemfile to `~> 2.0` or similar.

elmatou, 2016-03-29:
Ok, and know ?
- I let you move the drivers code to `Driver` class in specific module.
- I'll continue to test I2C and SPI with actual chips I ordered from adafruit.
- We need some new readme & wiki, I'll work on it.

- Do we want to have a suite of integration tests ?
it is pretty heavy to maintain, but as we are now, we do not assert PiPiper works
- what else to release version 3.0 ?

zsyed91, 2016-03-29:
Cool. I should have the modules done shortly so we can test everything once I push those. I also have some chips like a motion detector and various led's that I can test with as well.

Readme definitely needs updating and wiki as well. I would like to flush out the examples even more. The more code samples we can give, I think could be helpful to people.

Integration tests would be pretty cool if we could have them but I am not sure how we would implement them since they would have to run on the Pi itself. I think we can discuss that because it would be nice. It would reduce the burden of us testing manually. And yes it probably would be heavy to maintain.

I think we should be good to go for version 3.0 soon. In summary I think the following needs to be done under 3.0:

- Seperate the driver gems
  - I will finish creating the `Driver` class modules
- Update readme
- Update wiki
- Maybe add more/updated examples (this can be done after release IMO)
- Create integration tests (maybe, and can also be done after release if we decide to do it)
- For now, manually test the code on the Pi to make sure everything works

elmatou, 2016-04-03:
I just wrote some README for the gems. You can found them as gists here.
https://gist.github.com/elmatou/a4ce34818f887653a2252a358d9fb6c8
https://gist.github.com/elmatou/ae583d4589fbdeb4642ff370d9fc4efe
https://gist.github.com/elmatou/a3c9acbcea8456db56f0562da8bf2a1d

I can PR them or you do it with your next update.

zsyed91, 2016-04-03:
Cool, thanks for doing this. I will review your gists and then you can open them as PR's.
In the meantime, can you take a peak at https://github.com/PiPiper/sysfs/pull/4 before I merge it?