jwhitehorn/pi_piper #63: Trying to connect RC522 through SPI

YushengLi, 2016-03-02:
I am trying to use `pi_piper` to connect RC522 RFID reader through SPI, but I'm stuck right now, I think it's probably because I have only little knowledge of SPI, is there any tutorials or sample code I can refer to?

zsyed91, 2016-03-02:
@YushengLi I don't have much experience with SPI either but take a look at the following:

- https://github.com/jwhitehorn/pi_piper#spi
- https://github.com/jwhitehorn/pi_piper/blob/develop/examples/mcp3008_spi/mcp3008_spi.rb
- http://jeremyblythe.blogspot.com/2012/09/raspberry-pi-hardware-spi-analog-inputs.html

Hopefully one of those links should help you. There are other examples in the `/examples` directory of this repo. Let me know if you have any luck or if you find any other resources that help you. I would like to add them to our readme to help others.

Good luck :smile:

elmatou, 2016-03-02:
RC 522 seems to have I2C too. maybe you will have more chance that way.

zsyed91, 2016-10-18:
Hi @YushengLi did you get your project working via SPI?

YushengLi, 2016-10-18:
My friend dig into MFRC522's spec and eventually build a Gem for reading signal from MFRC522 using pi_piper with SPI interface.

Since I am not familiar with SPI connection, if you're interested in how he did it, you can take a look at his repo [atitan/MFRC522_Ruby](https://github.com/atitan/MFRC522_Ruby).