jwhitehorn/pi_piper #64: ERROR:pi_piper requires RubyGems version >= 2.0.0.

Damaotimlin, 2016-03-04:
Hi Folks,

I'm facing an error while I do
`sudo gem install pi_piper `

I return an error blow :
> ERROR:  Error installing pi_piper: pi_piper 
> requires RubyGems version >= 2.0.0. 
> Try 'gem update --system' to update RubyGems itself.

I use rvm and below is my ruby -v and gem -v
and do tried 'gem update --system' but still failed.

My ruby and gem is up to date:
> pi@cvproto01:~/pi-piper $ `which ruby`
> /home/pi/.rvm/rubies/ruby-2.2.1/bin/ruby
> pi@cvproto01:~/pi-piper $ `ruby -v`
> ruby 2.2.1p85 (2015-02-26 revision 49769) [armv7l-linux-eabihf]
> pi@cvproto01:~/pi-piper $ `which gem`
> /home/pi/.rvm/rubies/ruby-2.2.1/bin/gem
> pi@cvproto01:~/pi-piper $ `gem -v`
> 2.6.1

I did follow the instruction and install `libssl-dev`

And, I tried to install gem without `sudo`
`gem install pi_piper`
> Successfully installed pi_piper-2.0.0
> 1 gem installed
Guess what, it is successfully installed ?!

I was thinking I'm so smart to do no 'sudo'.
But the story always comes with HOWEVER 
However, I ran my ruby code, it returns:

> pi-piper-practice.rb:1:in `require': no such file to load -- pi_piper (LoadError)
> from pi-piper-practice.rb:1 

The pi-piper-practice.rb code listed below:

> require 'pi_piper'
> include PiPiper
> 
> 	## Examples
> 	## Example 1 - Simple switch
> 	puts "Press the switch to get start"
> 	PiPiper.watch :pin => 17, :invert => true do |pin|
> 		puts "Pin changed from #{pin.last_value} to #{pin.value}"
> 	end
>
> PiPiper.wait

I couldn't know what I am missing now. Please helps and thanks in advance !!!!!!

elmatou, 2016-03-04:
rvmsudo should do the trick.

rvm requires you install gems in "user space" meaning no sudo.
but pi_piper works (for now) only with sudo.

`rvmsudo` or `rbenv sudo` depending your ruby install allow to run with sudo gem installed in "user space"

Damaotimlin, 2016-03-04:
Hi @elmatou,

Thanks for your reply!

'rvmsudo' it works fine. Gem installed!
`rvmsudo gem install pi_piper`

> Warning: can not check `/etc/sudoers` for `secure_path`, falling back to call via `/usr/bin/env`, this breaks rules from `/etc/sudoers`. Run:
> 
>     export rvmsudo_secure_path=1
> 
> to avoid the warning, put it in shell initialization file to make it persistent.
> 
> In case there is no `secure_path` in `/etc/sudoers`. Run:
> 
>     export rvmsudo_secure_path=0
> to avoid the warning, put it in shell initialization file to make it persistent.
> Successfully installed pi_piper-2.0.0
> Parsing documentation for pi_piper-2.0.0
> Installing ri documentation for pi_piper-2.0.0
> Done installing documentation for pi_piper after 8 seconds
> 1 gem installed

However, when I run the practice again
`sudo ruby pi-piper-practice.rb`
It still return me

> pi-piper-practice.rb:1:in `require': no such file to load -- pi_piper (LoadError)
> 	from pi-piper-practice.rb:1

How to check if I installed the pi_piper correctly ?

Thank you again for help me collect the puzzles!

Damaotimlin, 2016-03-04:
Hi @elmatou,

I just found out two things and problem solved !

One is run 
`rmvsudo gem install pi_piper`

and run ruby file by

`rvmsudo ruby ~.rb`

All done, thanks !

closed 2016-03-04 by Damaotimlin

elmatou, 2016-03-04:
nice. but I don't think you need rvmsudo for installing the gem, only for running script.

Damaotimlin, 2016-03-05:
Hi @elmatou 

Thanks for your kindly helps!