jwhitehorn/pi_piper #67: Is Pi_Piper able to read DS18B20 heat sensor ?

Damaotimlin, 2016-03-08:
Hi Folks,

Pi_piper is great ! 

I'm wondering is Pi_Piper able to read DS18B20 heat sensor ? If yes, how to do it ?
If not, any recommendation gem or solutions ? 
My goal is to push the temperature data to Dashing dashboard. 

My code works fine run in terminal but seems not able to be fire in Sinatra(Dashing). 
```
def gettemp(id)

  filename = 'w1_slave'
  filepath = File.new("/sys/bus/w1/devices/#{id}/#{filename}", "w").path
  # “File” belongs to IO class, which contain I/O stream. http://ruby-doc.org/core-2.0.0/IO.html

  File.open(filepath) { |f|
    lines_arr = f.read.lines.to_a
    lines_arr.shift(1) # Let lines_arr takeout first object; second_line_s = lines_arr.to_s # Just for noted
    temp_arr = lines_arr.to_s.partition(/.t=/) # saperate the ["xx xx xx xx xx t=", "34.xxxx"]  
    @temp = temp_arr.last.to_f/1000
  }
  return @temp
end

# Use for startup the file
if __FILE__ == $0 then
  puts "Temp : #{gettemp('28-031581efbaff'})
end
```

Below is I'm trying to read the sensor in Dashing
> ~/dashing_project/jobs/sample.rb

```
# For reading temp/humidity sensor DHT11 gem
require 'dht-sensor-ffi'
# For reading soil remp sensor        
def get_soil_temp(id)
    filename = 'w1_slave'
    filepath = File.new("/sys/bus/w1/devices/#{id}/#{filename}", "w").path
    File.open(filepath) { |f|
        lines_arr = f.read.lines.to_a 
        lines_arr.shift(1) 
        temp_arr = lines_arr.to_s.partition(/.t=/) 
        @temp = temp_arr.last.to_f/1000
    }
    return @temp
end
soil_temp_sensor_id = '28-031581efbaff'

current_air_temp = 0
current_air_humid = 0
current_soil_temp = 0

SCHEDULER.every '5s' do
  last_air_temp = current_air_temp
  last_air_humid = current_air_humid
  sensor_val = DhtSensor.read(17, 11)
  current_air_temp = sensor_val.temp
  current_air_humid = sensor_val.humidity

  if __FILE__ == $0 then
    current_soil_temp = get_soil_temp(soil_temp_sensor_id)
  end

  send_event('air_temp', { current: current_air_temp, last: last_air_temp })
  send_event('air_humid', { current: current_air_humid, last: last_air_humid })
  send_event('soil_temp',  value: current_soil_temp)
end
```

Thank you in advance !

elmatou, 2016-03-08:
I wrote a gem for one wire devices. it is backed by the kernel driver (/sys/bus/w1/devices stuff)

I need to polish it and publish it. if you can wait until sunday, you could use it.

elmatou, 2016-03-08:
take a look at https://github.com/asapp/one_wire
we can discuss on it later.

Damaotimlin, 2016-03-08:
Hey @elmatou 
Great, thank you! I could wait and just let me know if you need some hand. Actually, I'm not good enough to do you some help, but I can test it after Sunday!

Looking forward to your Gem!!

elmatou, 2016-03-08:
You can check it now it should work for thermometer.

closed 2016-03-15 by Damaotimlin