jwhitehorn/pi_piper #68: Doesn't work on Raspbian JESSIE?

arnlen, 2016-03-13:
I try to use this gem but I don't manage to control any pin.
I have no error message, but nothing happens on my board.

The LED on pin 17 is on, since the boot of the raspberry. But:

```ruby
pin = PiPiper::Pin.new(:pin => 17, :direction => :out)
=> #<PiPiper::Pin:0x1ea4448 @pin=17, @direction=:out, @invert=false, @trigger=:both, @pull=0, @last_value=nil, @value=0>

pin.on
=> 1

pin.off
=> 1

pin.on?
=> false

pin.off?
=> true
```

Does anyone succeed to make it work on Raspbian Jessie + Ruby 2?

```
root@raspberrypi:~# ruby -v
ruby 2.1.5p273 (2014-11-13) [arm-linux-gnueabihf]
```

zsyed91, 2016-03-14:
Haven't gotten a chance to upgrade to jessie yet so I am not sure. It seems like the changes in jessie mean you don't need sudo access for GPIO anymore. But to be safe, did you try running your script or irb session using `rvmsudo`?

I can take a look but AFAIK `pin.on` and `pin.off` will always return 1 if it was successful. So the "state" is correct here, when you do `pin.off?`. Just to clarify, is this your workflow:

1. Led @ 17 is off
2. Instantiate pin with `PiPiper` and do `pin.on`.
3. Led @ 17 is now on
4. `pin.off`
5. Led @ 17 is still on?

elmatou, 2016-03-14:
tell us also which version of PiPiper you are using.

arnlen, 2016-03-14:
Here is what I notice:
- connected to 3.3V pins (pin #1 or #17), the LED if on when the board is powered on.
- connected to another pin (GPIO 17 on pin #11), the LED is off on startup.

Then:

```
# same as rvmsudo but for rbenv
rbenv sudo irb

irb(main):001:0> require 'pi_piper'
=> true
irb(main):002:0> include PiPiper
=> Object

irb(main):005:0* pin = PiPiper::Pin.new(:pin => 13, :direction => :out)
=> #<PiPiper::Pin:0x5665da18 @pin=13, @direction=:out, @invert=false, @trigger=:both, @pull=0, @last_value=nil, @value=0>
# LED is off

irb(main):006:0> pin.on?
=> false

irb(main):007:0> pin.on
=> 1
# LED is still off

irb(main):008:0> pin.on?
=> false
```

arnlen, 2016-03-14:
If I try to use pi_piper without `sudo`:

```ruby
irb(main):004:0* pin = PiPiper::Pin.new(:pin => 11, :direction => :out)

bcm2835_init: Unable to open /dev/mem: Permission denied
Errno::EACCES: Permission denied @ rb_sysopen - /sys/class/gpio/gpio11/direction
	from /home/pi/.rbenv/versions/2.3.0/lib/ruby/gems/2.3.0/gems/pi_piper-2.0.0/lib/pi_piper/bcm2835.rb:61:in `write'
	from /home/pi/.rbenv/versions/2.3.0/lib/ruby/gems/2.3.0/gems/pi_piper-2.0.0/lib/pi_piper/bcm2835.rb:61:in `pin_direction'
	from /home/pi/.rbenv/versions/2.3.0/lib/ruby/gems/2.3.0/gems/pi_piper-2.0.0/lib/pi_piper/bcm2835.rb:45:in `pin_output'
	from /home/pi/.rbenv/versions/2.3.0/lib/ruby/gems/2.3.0/gems/pi_piper-2.0.0/lib/pi_piper/pin.rb:56:in `initialize'
	from (irb):4:in `new'
	from (irb):4
	from /home/pi/.rbenv/versions/2.3.0/bin/irb:11:in `<main>'
irb(main):005:0>
```

arnlen, 2016-03-14:
It works!
My bad, I was misunderstanding the pin number: I though the pin number was the number in the order of the GPIO (from 1 to 40), but in fact I understand there is no relation between the position of the pin, and it's number.

The GPIO #17 pin if on the place 11, but it's still called "17".

Conclusion:
- still works on Jessie
- still requires `sudo`

Thanks for your help! :)

closed 2016-03-14 by arnlen

zsyed91, 2016-03-14:
Ah yes, I was thinking that might be it. I will update the documentation to make this clearer. I noticed that others have had that same misunderstanding. I am glad that it works for you though!