jwhitehorn/pi_piper #69: Is Pi_Piper ready for RPi 3 ?

Damaotimlin, 2016-03-15:
I'm thinking to get a new RPi 3 because inboard wifi, but it is upgrade to bcm2387. I'm wondering is Pi_Piper can be use in Pi 3 ?

zsyed91, 2016-03-15:
Haven't been able to get one yet to test one, completely sold out everywhere by me :smile: 
However, it should work. The only thing you would need to do possibly, is to keep an eye out here: http://www.airspayce.com/mikem/bcm2835/ and if a new version gets released, just swap out the `.so` file. We are currently working on separating out the drivers from the gem itself to accommodate these kinds of issues. So hopefully once that is released soon then all you would need to do is use a different driver with the `PiPiper` gem.

Let me know once you get the RPi 3 whether it works for you or not. This is definitely a priority but since the board is new, and relatively difficult to obtain I am sure most people for now will be using the 2.

Damaotimlin, 2016-03-15:
I'll try after I get a new Pi 3 !

elmatou, 2016-03-16:
the sysfs driver will work without problem, the bcm2835 driver will need some library update as @zsyed91 mentioned.

Damaotimlin, 2016-09-02:
Sorry for the late update. I tried and it works prefect on Pi 3. So, I'm closed this issue! P.S. I use bcm2835 library.

closed 2016-09-02 by Damaotimlin