jwhitehorn/pi_piper #75: Output pin turns back on after exit

booksmart00, 2016-04-28:
I was testing out the functionalities of the library with some LEDs and I noticed that even if it is explicitly turned off, when the program exits, the pin turns back on at a slightly lower power.

booksmart00, 2016-04-28:
As it turns out, this happens even when the Pi is turned off (and is still plugged in to a power source). It also happens regardless of which library I use (I also tested Python's RPi.GPIO). It appears to have something to do with the Pi itself. My apologies.

elmatou, 2016-04-30:
Without any pull up or down, the value of the pull can't be asserted, it is pretty common, and every circuit should have a pull-up or sink resistor on the GPIO.

closed 2016-10-18 by zsyed91