jwhitehorn/pi_piper #77: Yet Another Problem

riolibrary, 2016-06-13:
After doing Sudo ruby xxxx.rb
/usr/uibl ruibylgews/1.9.1/ (Errno: :EBUSY) e Device or resource busy /sys/class/gpio/export r from piper-2.0.0/u b:65:in export from /usr/lib/ piper-2.e.e/uib/pi piper/bca283 b:34:in pin input. piper.2.0.0/uib/pi piper/pin.rb: sin initialize from /usr/lib/ruby/gems/1.9.1/gems/pi piper 2.0.0/uib/pi piper.rb115:in from /usr/lib/ruby/gems/1.9.1/gems/pi piper-2,0.0/uib/pi piper,rb1151in new block in Watch 

Not completely correct.
Sorry😀

zsyed91, 2016-06-15:
instead of `sudo ruby xxx.rb` please run `rvmsudo ruby xxx.rb`.

Check out [https://rvm.io/integration/sudo](https://rvm.io/integration/sudo) for a detailed explanation. `sudo` does not pass the session to the script which is why `rvmsudo` is needed. Please try that and let me know if it works for you.

riolibrary, 2016-06-17:
I don't know how to use RvmSudo

zsyed91, 2016-06-20:
Well, its pretty simple:

1. [Install rvm](https://rvm.io/rvm/install)
2. Install a ruby via `rvm install ruby-2.1.0` or if you want latest, `rvm install ruby-2.3.0`.
3. Run `rvmsudo ruby xxx.rb`

riolibrary, 2016-06-23:
That is simpler than I expected but do I have to reinstall ruby? I have it installed already and it took me forever to re update everything.

zsyed91, 2016-06-25:
I don't believe so, if you wanted to use the system ruby or whichever you have installed already. In that case, simply try running step 1 and step 3.

zsyed91, 2016-07-27:
@riolibrary Did you get your project working?

riolibrary, 2016-07-30:
#NO
@zsyed91 I tried to run

`rvmsudo ruby xxx.rb` and it gave the same error except with something about a secure path
-----
Please help, I have been with this project for a couple months now and would like to finish it

elmatou, 2016-07-31:
from the few line you posted I can understand that you run with a linux permission issue, which is not specific to PiPiper.

Rasbian requires that gpio are accessed by an user wit root privileges (which is not really true with jessy based raspian, but for clarity we will stick to that assomption)

@riolibrary, could you copy/paste an extensive backtrace showing your bug.
Also we need to know which ruby you installed and how you did it.

maraaaa, 2016-09-26:
@riolibrary did you ever get this issue resolved?  If so, what was the fix?

If not can we try a few things?

* execute a `sudo -i` which will drop you to a root shell, then as root, execute your `ruby xxxx.rb` command.
* execute `sudo -E ruby xxxx.rb`, the `-E` flag tells sudo to "preserve environment".  As @zsyed91 points out, it's likely an environment issue.  Since you're not using RVM to manage Ruby, I don't think `rvmsudo` is the solution though.  (Though, if I may ask, why aren't you using RVM?)
* According to the above [linked post](https://rvm.io/integration/sudo): `the user that is invoking sudo must have export rvmsudo_secure_path=0`, and your comment about a secure path error,  I would wager that this is not set before running `rvmsudo`.  Did you set this?

Finally, as @elmatou points out, we'll need some more logging to better help you debug.  If you can post a full backtrace of your errors it would be immensely helpful.