jwhitehorn/pi_piper #78: Sorry if it's a noob question...

jvalencia80, 2016-07-26:
but, why can't I see the `bcm2835.rb` file here on github?

zsyed91, 2016-07-27:
So we are in the process of rewriting the library a bit. Its actually basically ready for release. I just want to test it a bit more to make sure the release goes smoothly since it breaks backwards compatibility.

You're most likely looking at the `development` branch. If you change the branch to `master` you'll see the currently released code.

maraaaa, 2016-09-26:
@zsyed91 how's the testing going?  Anything I can do to help?  

Would you recommend using the `development` branch at this time?

Thanks!

elmatou, 2016-09-27:
I don't think you should use it for now. The rewrite is based on a plugin gem approach for the drivers.
And the drivers where not yet reviewed by @zsyed91 and released.
https://github.com/PiPiper/bcm2835
https://github.com/PiPiper/sysfs

Each of the drivers use a specific way to handle the gpio (file based access, or the C++ library that access /dev/mem). Sadly this work is stalled for several month now.

zsyed91, 2016-09-27:
Apologies, I've been working on it on the side. I think we're ready for a
release. I'll try and get this out by the weekend. I haven't forgotten
about this :)

I think our approach will work perfectly in this scenario once we release.
Nice to hear from you!

elmatou, 2016-09-27:
@zsyed91 good to hear from you too. Ask me for anything when you'll be on it..

zsyed91, 2016-09-27:
Sure will do. I'll ping you tomorrow morning, it's very late here :)

In the mean time, did you do any real world testing with the new changes? I
tested it with a couple sensors and it seems pretty stable.

elmatou, 2016-09-27:
yes, Pin works fine for me with sysfs, but won't trigger smoothly with bcm2835 C++ lib.
I have to double check SPI and I2C

elmatou, 2016-09-27:
we really should have a policy about integration testing...

zsyed91, 2016-10-18:
Regarding integration testing, I absolutely agree. There must be a better way than physically hooking everything up 😄 . I think let's do this release and then think of some possible ways we can get integration testing without needing physical hardware. It will probably require a lot of effort but if at all possible will make things way more robust.