jwhitehorn/pi_piper #79: Can't read correct values from a rpi B 1

jvalencia80, 2016-07-31:
I'm using pin 4 (gpio 4) with this simple code:

`require "pi_piper"

pin = PiPiper::Pin.new(:pin => 4, :direction => :in)

loop do
        puts pin.value
        sleep 0.1
end`

I use a jumper and connect it to gpio4 and 3.3V (top-left).

It always shows a 0. If I change the jumper to GND (bottom-left), I also get a 0.

Reading the values from /sys/..., it shows correctly a 1 or 0, but the pi_piper script will always show a 0.

What's wrong?

See the attached screenshot
[image]

elmatou, 2016-07-31:
just update the value in memory with `pin.read`.
I'm not a huge fan of this implementation, but we don't want to change it for compatibility sake.

jvalencia80, 2016-07-31:
mmm I see now, it's cached. 

Do you think something may break if you alias `value `and `read`???

Who needs it to be cached? Why would you want to read an "old" cached value that may be incorrect?

elmatou, 2016-07-31:
Yep, `value` is used against last_value to 'detect' changes.
If this behaviour really annoy you, I would monkey patch PiPiper::Pin with
```
    def value
      @value = read
    end
```

jvalencia80, 2016-07-31:
Ah, now I understand why they chose this implementation. It's a bit inconsistent, as you said, because it's not clear what method should be used for getting an updated value without looking into the source code.

I think I have a solution to "fix" (let's call it that way) this inconsistency:

```
def changed?
	@changed
end

def value
	val = Platform.driver.pin_read(@pin)
	@changed = (@last_value != val)
	@last_value = val
	invert ? (val ^ 1) : val
end

alias_method :read, :value
```

I think nothing breaks with that change, and the alias is there for backward compatibility. `@value` is not used anymore. `changed?` remains constant betweens reads, as the original code.

Also, if you call the original `changed?` method before calling `read`, `last_value` will be **nil** (not initialized) and `changed?` will return **true** (nil != int, false positive). My idea fixes that because `@changed` defaults to **nil** (false).

Just my 2 cents.

zsyed91, 2016-10-18:
While I agree that it may be a bit inconsistent. We have this method already `read`.

https://github.com/jwhitehorn/pi_piper/blob/develop/lib/pi_piper/pin.rb#L97.

We can certainly update the docs to make this more evident.