jwhitehorn/pi_piper #80: One-wire monitoring with PiPiper

maraaaa, 2016-09-26:
Hello,

I am new to the Raspberry Pi and have been looking to expand my Ruby expertise beyond Sinatra apps and Chef code.

PiPiper seems to be the most mature (and only still active) Ruby library, so it seems like the natural jumping off point.

Looking at the documentation and the code, it doesn't look like there's a One-wire bus interface.  Can you confirm this?  Am I just looking in the wrong place?

I think I can take [pin.rb](https://github.com/jwhitehorn/pi_piper/blob/develop/lib/pi_piper/pin.rb) and kind of extend what's happening there to be able to monitor my one-wire sensors?

Essentially what I think is missing (please correct me anywhere I'm mistaken):
* A method to read every `/sys/bus/w1/devices/28-XXXXX/w1_slave` interface
* A method to parse the data into (<device>, <bus>, <temp>) (since w1_slave is two lines of output) (maybe appropriate to include a timestamp?)
* A method to determine if the value has changed since last poll.

Am I thinking along the right lines here?  Any thoughts?  Would one-wire monitoring be a feature that would be appropriate/welcome in PiPiper?

Thanks!

elmatou, 2016-09-27:
PiPiper rewrite would be able to handle 1-Wire devices.
for now I wrote : https://github.com/asapp/one_wire which do a good enough job.

zsyed91, 2016-09-27:
Yup I think this validates our rewrite effort! As mentioned in the other
issue, I'm going to get this pushed along. Hoping to release this weekend.
I think the changes are pretty stable.