jwhitehorn/pi_piper #81: Why is lib/pi_piper/driver.rb empty method definitions?

maraaaa, 2016-10-18:
Hello,

I'm trying to understand how this code works, and I see in `lib/pi_piper/pin.rb` you call `PiPiper.driver.pin_write`  - https://github.com/jwhitehorn/pi_piper/blob/develop/lib/pi_piper/pin.rb#L61

In `lib/pi_piper.rb` you seem to initialize `PiPiper.driver` as an instance of `PiPiper::Driver`  https://github.com/jwhitehorn/pi_piper/blob/develop/lib/pi_piper.rb#L19-L32

What I don't understand, is why is `lib/pi_piper/driver.rb` full of empty method definitions?  Where do you actually do the writing to filehandles?  

In previous versions you actually wrote to `/sys/class/gpio/gpio#{pin}/value` - https://github.com/jwhitehorn/pi_piper/blob/master/lib/pi_piper/bcm2835.rb#L40

What am I missing here?  How does this know to write to the correct location?

Thanks!

zsyed91, 2016-10-18:
Hi, we are about to release `3.0` where we pulled out the library so it can be driver-agnostic.

We will default to use the `sysfs` driver but currently we have 2 implementations ready.

1. https://github.com/PiPiper/sysfs/blob/master/lib/pi_piper/sysfs/sysfs.rb
2. https://github.com/PiPiper/bcm2835/blob/master/lib/pi_piper/bcm2835/bcm2835.rb

We have empty method definitions because we wanted to have an interface in which we can spin out other drivers while making sure that they follow what we have defined.

mrambossek, 2019-06-01:
hello good people. any updates on this?

elmatou, 2019-06-03:
The gel is on hold/abandoned, by the maintainers. Everything should be easy to get working.
I plan to spin out the code and start a new project based on release 3.0 but I lack time for now.
If you are willing to work on this, I may be able to help you.